dl9rdz/rdz_ttgo_sonde #435: RDZTTGO APRS Reporting Issue

obi7zik, 2024-04-19:
I've noticed a problem with the RDZTTGO APRS reporting. Even after selecting option 0 to disable location reporting, data continues to be transmitted. Could you look into this. ( firmware version devel20240316-B17) device ttgo T-Beam.

73-de-LZ, 2024-04-20:
Checked, when APRS location reporting=0 ,save then reboot devel20240316-B17 working as expected.

obi7zik, 2024-04-20:
> Checked, when APRS location reporting=0, save then reboot devel20240316-B17 is working as expected.

Despite several attempts, I couldn't achieve success. Although option 1, which entails reporting a fixed location, is functional, I'm utilizing a T-Beam 1.1 board with a neo M8 module, and my preference is to have no location reported at all.

hr-ru, 2024-04-20:
Yes I think there is something I need to fix. 
Currently my Travis CI plan for automated builds has expired, so I first need to get that part running again :)

73-de-LZ, 2024-04-20:
APRS and sondehub locations are different things, just to be sure we are at the same wavelenght?

obi7zik, 2024-04-21:
> APRS and sondehub locations are different things, just to be sure we are at the same wavelenght?

The location reporting of SondeHub is functioning as expected without any issues, while APRS report the exact GPS location on radiosondy.info with the "0" option selected.

dl9rdz, 2024-08-12:
Thanks for reporting this issue, this has been fixed now (in the latest commit).
Will be in the new IDF branch in the next dev version [note: no OTA available from old to new IDF)

closed 2024-08-12 by dl9rdz