dl9rdz/rdz_ttgo_sonde #439: Temperature and humidity missing in MQTT message

william67, 2024-05-19:
Temperature and humidity missing in MQTT message:

{
   "lat": 51.98738,
   "lon": 5.16816,
   "alt": 28719.5,
   "vs": 5.2,
   "hs": 8.5,
   "climb": 5.2,
   "speed": 8.5,
   "dir": 282.9,
   "type": "RS41",
   "id": "V1620877",
   "ser": "V1620877",
   "frame": 6507,
   "vframe": 6507,
   "time": 1716080679,
   "sats": 10,
   "freq": 403.9,
   "rssi": 237,
   "afc": 183,
   "launchKT": 0,
   "burstKT": 30600,
   "countKT": 0,
   "crefKT": 0,
   "launchsite": "DeBilt",
   "res": 0,
   "batt": 2.6
}

Steps to reproduce the behavior:
- have installed dzTTGOserver devel20240514 on LilyGo T3_V1.61 LoRa32 433MHz
- configured WIFI
- activated debilt and change frequency to 403.900
- configured MQTT and enable log incoming messages in an MQTT logger
- not other changes in settings or so
- wait for data, see above message

Data packets are coming in the MQTT logger but Temperature and humidity is missing for some reason.

Hardware & software
dzTTGOserver devel20240514 on LilyGo T3_V1.61 LoRa32 433MHz

Done another test last year, also then the Temperature and humidity date was missing
Others (https://s1.radiosondy.info/) seems to have temperature and humidity data.

mycarda, 2024-07-13:
I am running devel20240521 and I can see the temperature and humidity in the MQTT message. According to the code, it looks like the MQTT message will only contain temperature and humidity information if the device has calculated a valid temperature and humidity but this depends on receiving enough calibration frames first. I generally have to wait a few minutes before valid temperature and humidity appears on my device. After a few minutes, my MQTT packed looks like this.

rdzDev/packet {
  "lat": 51.29696,
  "lon": -1.71817,
  "alt": 14337.7,
  "vs": -6.2,
  "hs": 10.4,
  "climb": -6.2,
  "speed": 10.4,
  "dir": 58.7,
  "temp": -44.8,
  "humidity": 1.0,
  "type":"RS41-SG",
  "id": "W1010387",
  "ser": "W1010387",
  "frame": 5039,
  "vframe": 5039,
  "time": 1720765653,
  "sats": 10,
  "freq": 404.40,
  "rssi": 207,
  "afc": 4760,
  "launchKT": 65535,
  "burstKT": 30600,
  "countKT": 65535,
  "crefKT": 4998,
  "launchsite": "Larkhill",
  "res": 0,
  "batt": 2.6
}

@william67 did your MQTT packets fail to show the temperature and humidity even after several minutes? It sometimes takes a long time before I can see the temperature and humidity if I have a weak signal and I am unlucky to miss a critical frame containing calibration data.

mycarda, 2024-07-16:
In my last post, I suggested you might not see the temperature and humidity for several minutes. It can be a lot longer than that. This morning, the first packet I received was

05:48:22
```rdzDev/packet {"lat": 51.15005,"lon": -1.73372,"alt": 7507.4,"vs": 4.5,"hs": 13.8,"climb": 4.5,"speed": 13.8,"dir": 196.1,"type":"RS41","id": "W1010613","ser": "W1010613","frame": 2588,"vframe": 2588,"time": 1721108902,"sats": 8,"freq": 404.40,"rssi": 223,"afc": 4638,"launchKT": 0,"burstKT": 0,"countKT": 0,"crefKT": 0,"launchsite": "Larkhill","res": 0,"batt": 2.9}```

The signal was poor so I was missing most of the packets. It was not until much later that I had received enough packets to populate the calibration information for the temperature.

05:54:37
```rdzDev/packet {"lat": 51.09676,"lon": -1.76886,"alt": 9765.3,"vs": 5.6,"hs": 15.3,"climb": 5.6,"speed": 15.3,"dir": 148.5,"temp": -40.4,"type":"RS41-SG","id": "W1010613","ser": "W1010613","frame": 2963,"vframe": 2963,"time": 1721109277,"sats": 8,"freq": 404.40,"rssi": 218,"afc": 4638,"launchKT": 65535,"burstKT": 30600,"countKT": 65535,"crefKT": 2754,"launchsite": "Larkhill","res": 0,"batt": 2.8}```

It was nearly another two minutes before I had enough calibration packets to calculate the humidity.

05:56:23
```rdzDev/packet {"lat": 51.08705,"lon": -1.75681,"alt": 10318.6,"vs": 5.7,"hs": 10.5,"climb": 5.7,"speed": 10.5,"dir": 126.3,"temp": -40.4,"humidity": 2.0,"type":"RS41-SG","id": "W1010613","ser": "W1010613","frame": 3069,"vframe": 3069,"time": 1721109383,"sats": 8,"freq": 404.40,"rssi": 224,"afc": 4760,"launchKT": 65535,"burstKT": 30600,"countKT": 65535,"crefKT": 3060,"launchsite": "Larkhill","res": 0,"batt": 2.8}```

So, I think it is not true that the temperature and humidity is missing from MQTT packets, but it is true that sometimes it takes a long time before they appear in the MQTT packet.

closed 2024-09-04 by dl9rdz

william67, 2024-09-04:
Yes, this issue can be closed. 
I was not aware of the fact that before calculating temperature and humidity first some calibaration data needs to be collected.
Due to poor reception / antenna position I am missing a lot of frames, so that has to be solved first.
Regards, William

william67, 2024-09-04:
@mycarda Thanks for helping me with this, I hadn't seen your message before because I was on holiday.