dl9rdz/rdz_ttgo_sonde #440: DFM17->DFM09 mis-idenficiation

darksidelemm, 2024-05-20:
**Describe the bug**
DFM17 radiosondes being mis-identified as DFM09 sondes due to the way we are currently doing identification using the maximum channel number. I'm pretty sure rdz_ttgo_sonde is doing pretty much the same thing as the RS decoders in this respect.

**To Reproduce**
Receive a 'new' NWS DFM17 sonde... something has changed in their configurations recently resulting in them being identified as DFM17s, as their max channel count is 0xA.

**Expected behavior**
These sondes get identified properly!

**Additional context**
Copied from here: https://github.com/projecthorus/sondehub-infra/issues/141

> We're beginning to see many DFM17 launches mis-identified as DFM09 by the various decoders out there (all seem to be behaving consistently... I think).
> 
> The method that we currently use to identify DFM sub-types is known to be flawed, as changes in sensor configuration can result in the maximum channel ID field that we use for identification to change.
> 
> We have reliable information that DFM09 sondes have not been manufactured since early 2023, and that the highest DFM09 serial number is around the 22033000 area (presumably programmed in 2022).
> 
> So, the consequences of a mis-detection are:
> 
> Incorrect identification of subtype on the receiving software and tracker
> Incorrect calculation of sensor temperature due to using a different Rf value (refer line 862 in dfm09mod.c).

What we've been doing:
* SondeHub is now converting telemetry which has a subtype of DFM09 and a serial > 22033000 to DFM17.
* @rs1729 has recently made changes in dfm09mod.c which will convert DFM09 detections to DFM17 for serial numbers higher than 23000000. They are also using the signal polarity to confirm if a sonde is DFM09 - I'm unsure if this approach can be implemented in rdz_ttgo_sonde.

dl9rdz, 2024-05-21:
Yes, polarity is available and should always be reliable (unlike RS where it might depend on the receiver?).
So I assume that the simplest approach is to just use the polarity to decided if it is a DFM09 or a DFM17 if max. channel is 0xA.
I added this functionality in devel20240521.

dl9rdz, 2024-08-12:
Closing this issue as I assume it is resolved.

closed 2024-08-12 by dl9rdz

dl9rdz, 2024-08-19:
There is one more thing, more about sondehub:
Due to a lot of breaking changing (new Espressif framework version, new partitioning scheme) and, due to this, the lack of OTA upgrade/downgrade possibility between the version, I decided to use a different naming scheme
develXX and masterXX are the old version
devXX and mainXX are the new version.
As sondehub seems to use the version number to block old version from uploading potentially incorrect DFM data, could you update that part to accept the new version strings?

darksidelemm, 2024-08-19:
Our AWS Lamda guru is currently on a 4WD trip across the Simpson Desert, so this might not happen for a little while...

darksidelemm, 2024-09-08:
This should be fixed with https://github.com/projecthorus/sondehub-infra/pull/146