dl9rdz/rdz_ttgo_sonde #449: PS15 recognized as DFM06P in latest dev20240817-C2

73-de-LZ, 2024-08-18:
[image]

If helpfull: PS15 recognized as DFM06P in latest dev20240817-C2

73!

dl9rdz, 2024-08-19:
Thanks for reporting this. This is actually two issues.
(a) sondehub does not handle my new version handling scheme yet
(b) the software does not handle PS15. Indeed seems like I did not implemented that yet as I never had a PS15 for testing. Telling PS15 and DFM6P apart seems to be a bit trickier, but when in doubt then its way more likely a PS15. I'll add something in the next version in this regard

73-de-LZ, 2024-08-19:
Also i think QRG import from sondehub doesn't work. Also RS92 is decoded, but data don't reach to the trackers, sondehyb and SQ6KXY

73-de-LZ, 2024-08-19:
Found something in the old versions, i hope it will not pop up again in the newest. In about sreen:
`RS92 RINEX eph state: Disabled (no RS92 in QRG list or Wifi mode not 3)`
I think it should be "or Wifi mode = 2 AP", as i always use mode=1, otherwise sudden after power failure, while the rooter boots up, RDZ goes in AP mode. The new feature 4=noscan, would be more useful if 4=static IP in my opinion

dl9rdz, 2024-08-20:
Yes QRG import is disabled in the current version (sorry, the only place I documented this was in a telegram channel, not on the repository), this will be back in the next version.

The rinex download only works if there is a connection on startup. Downloading and uncompressing the file takes some memory, so the initially idea was to do this before any other tasks run, in order to not risk running out of memory. Possibly, I should change this, maybe just temporarily disable the decoder while the download is being made. That would allow enabling RS92 while the device is running.

Regarding modes, you are right, mode 1/3/4 would all work equally in recent versions (I think this was different in earlier versions where mode 1 did not wait at all, but now they all initially wait for a moment to connect, and if successfully (with some time bound) the download is triggered.

I have seen IP=0.0.0.0 once but was not able to reproduce this so far, but thanks for reporting this. I guess this needs to be investigated.

14ri004, 2024-08-24:
hello hansi, I also have the connection problem with the new Dev2 version, (wifi mode =1), it sometimes tells me 0.0.0.0 as ip, so (reboot), it did not do it with dev1, it automatically switched from my Box to my 4G router in the car, and this every time I leave the house, is there a solution?, cdt