dl9rdz/rdz_ttgo_sonde #451: dev20240817-C2-full.bin

14ri004, 2024-08-22:
hello, firm dev20240817-C2-full.bin and T-BEAM 1.1 with ili9225, when starting the T-BEAM the information is too large

[image]

dl9rdz, 2024-08-22:
This is a configuration issue.
I assume this is a TTGO v2.1_1.6 board or very similar? In that case currently the integrated SD card slot is configured by default. I assume your large display uses some of those pins. In that case you need to disable the SD card (set all pins in the "SD card" config to -1) (or, alternatively, if you want to use the SD card, re-wire your display to different, unused pins).

The next version will include a config check that displays a warning when configuring some pins twice.

14ri004, 2024-08-22:
This is the T-BEAM 1.2 with AXP2101 and ili9225 screen, the settings are original and SD card not activated

[image]

dl9rdz, 2024-08-22:
Indeed, my fault. I'll make an update, there was a commit that did not make it into the repo...

14ri004, 2024-08-22:
I did the update to dev20240822-C3-full.bin and the result is identical

closed 2024-08-23 by 14ri004

hr-ru, 2024-08-23:
Great to see this is fixed. 

Let me add (in case of other people seeing the same problem and find this discussion):
The problem is not in the core firmware, but in the (separate) font partition with font data. So you need to either
- fully re-program the TTGO or
- re-upload the font partition (with platformio command line "pio run -t uploadfonts")
The over-the-air update mechanism currently does not update the font partition (that might be a TODO for the future as well)