dl9rdz/rdz_ttgo_sonde #453: screen saver malfunction

14ri004, 2024-08-25:
Hello hansi, I just saw that the screen saver does not work on the Dev2, I put 10s and the screen turns off fine, but it does not turn back on when decoding a probe, it remains black, with the Dev1 it works perfectly, any idea ??

[image]

dl9rdz, 2024-08-25:
Are you sure it works this way with dev1? Nothing has changed in the code for that part.

If you want the display to turn off after 10s if no signal is received, you should put the value "102" (10 seconds * 10 + 2 for only use screen safer if no signal is received).

The comment in the config tab is not fully accurate, just "0" is off, anything like XXX0 and XXX1 is treated as "always active, after XXX seconds), and XXX2 is "use screensaver if nothing is received, after XXX seconds.

So "10" in fact means the screen safer will turn the screen of after 1 second, in any case (if a signal is received or not). Actually this 1 second is in fact 0 seconds, as there is a 1 second delay when the display is activated, so it will indeed be disabled right away after begin enabled.    AFAIK this has always been this way.

(But ok, this can be enhanced, so that x seconds are actually x seconds and not x-1 seconds, and better documentation)

14ri004, 2024-08-25:
Hello, it actually works with 102, the screen turns off and turns on when decoding, with the right information it's immediately better, thank you

closed 2024-08-26 by 14ri004