dl9rdz/rdz_ttgo_sonde #456: tracking of DFM17

14ri004, 2024-09-03:
Hello, no tracking of DFM17 on Sondehub, decoding works but is not visible on Sondehub.

State: Connected: Waiting for ACK from server. Last upload start: -1 s ago Last reply: {"message": "some or all payloads could not be processed", "errors": [{"error_message": "rdzTTGOsonde branch and version was unable to be determined. We are unsure if this version handles DFM probes correctly. Please update to master 0.9.3, devel20230427 or later", "payload": {"software_name": "rdzTTGOsonde", "software_version": "dev20240901gps"

[image]

dl9rdz, 2024-09-03:
I am aware of this issue. sondehub checks the version number, and fails to do so with the "new" (dev instead of devel) version numbers. This should be fixed at the sondehub side but that will take some while (I am in contact with them already), so next version will temporarily send "fake" version numbers as a temporary workaround.

14ri004, 2024-09-03:
OK, that was to let you know about the problem, THANK YOU

closed 2024-09-03 by 14ri004