dl9rdz/rdz_ttgo_sonde #459: sending M10 data does not work either

14ri004, 2024-09-06:
Hello, for information, this morning I decoded a M10 901-2-11018 well, but invisible on sondehub, I started the SDR and the probe appeared well on Sondehub, sending data like the DFM17 does not work either with the M10

dl9rdz, 2024-09-06:
version number?

If the devXX, any sondehub status information on the web interface?

14ri004, 2024-09-06:
rdzTTGOsonde version dev20240905

dl9rdz, 2024-09-06:
What is on the status info? Any error message?

[image]

14ri004, 2024-09-06:
I just tested on the M10 210-2-12462, the decoding and data are sent well with this probe, and this morning on another M10 nothing is displayed on sondehub, I don't understand why, so some probes must be blocked or locked??

closed 2024-09-07 by 14ri004