dl9rdz/rdz_ttgo_sonde #460: invalid json being submitted to sondehub - `inf` in the humidity field.

TheSkorm, 2024-09-07:
**Describe the bug**
JSON sent to sondehub includes invalid json. Specifically in the humidity field we are receiving an `inf` value.

**To Reproduce**
Steps to reproduce the behavior:
- W1310471 radiosonde had a sensor fault
- TTGO stopped showing up on SondeHub

**Expected behavior**
Data ends up in SondeHub

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

SondeHub logs show:
```json
    {
        "software_name": "rdzTTGOsonde",
        "software_version": "master_v0.9.5",
        "uploader_callsign": "KF5LJD",
        "time_received": "2024-09-07T00:49:01.000Z",
        "manufacturer": "Vaisala",
        "serial": "W1310471",
        "datetime": "2024-09-07T00:49:18.000Z",
        "lat": 32.74656,
        "lon": -97.13464,
        "alt": 35540.33203,
        "frequency": 404.800,
        "vel_h": 4.67977,
        "vel_v": 8.47726,
        "heading": 257.74380,
        "rssi": -106.0,
        "frame": 8248,
        "type": "RS41",
        "sats": 9,
        "subtype": "RS41-NG",
        "temp": -251.7,
        "humidity": inf,
        "burst_timer": 65535,
        "batt": 2.60,
        "uploader_position": [
            33.280400,
            -96.988700,
            690
        ]
    },
```

**Version and specific configuration:**
Seems to be these versions:
[image]

**Additional context**
meow.

dl9rdz, 2024-09-07:
Good observation ... Will be fixed in the next version.

darksidelemm, 2024-09-07:
While it might not be the most ideal approach, the RS decoders used in auto_rx handle these kinds of failures with humidity by clipping the value to either 100 or 0 (depending on the fault).

mycarda, 2024-10-03:
Is the best approach to not send the temperature and humidity values if they are out of range? We don't send temperature and humidity data until we received enough frames to calculate a valid temperature and humidity, so if we know the values are out of range, just don't send data we know is wrong.