dl9rdz/rdz_ttgo_sonde #461: Automatic chase mode uploading fixed position in 0.9.5. No

rpratt20, 2024-09-09:
**Describe the bug*
I did over the air update from 0.9.4 to 0.9.5 on a Lillygo Lora board. After the update, sondehub posts both my fixed coordinates and my gps coordinates  when in auto chase mode.  See image.  I have rebooted several times and find I must select option 1 fixed to not show the chasecar gps.  I also noted that the board identification page no longer identifies the board as it did prior to update. See second image.

[image]
[image]

rpratt20, 2024-09-27:
I’m noticing some more information on this. I have option 1 fixed position for station and 3 qrg launch locations. 1 launching station shows my home coordinates in live.json. The other 2 stations report gps coordinates including hdop information. Live gps position continues to report to Chasemapper but Sondehub is showing fixed location while fixed option is selected.

hb9evw, 2025-05-02:
Please fix this issue which wasn't present in 0.94! It is causing unnecessary displays of chase cars instead of fixed receiver position icons on sondehub. It is quite annoying for us who want to know when a person is really chaasing.