dl9rdz/rdz_ttgo_sonde #468: Website does not work on iOS / iPadOS 18

cixio, 2024-10-01:
**Describe the bug**
On iOS 18 / iPad OS 18 when accessing the website via IP-Adress Safari shows "cannot parse response"

**To Reproduce**
Connect to TTGO, open website in Safari on iOS 18

**Expected behavior**
Website should be shown like in other browsers. No problems with Safari on MacOS or other browsers. No problem with chrome on iOS / iPadOS. Seems to be a safari problem on iPhone / iPad.

**Version and specific configuration:**
 - TTGO LORA32 v2.1_1.6 (0.9" OLED@21,22)
 - dev20240924

**Additional context**
When disabling / removing the template processor in source code there are no problems.

dl9rdz, 2024-10-01:
I don't have a iPad or iPhone for testing this...

I assume you are referring to the String processor() function?
Can you try simply removing the template processing for the autodetect info, e.g. change
  if (var == "AUTODETECT_INFO") {
to something else like
  if (var == "XXXREMOVED") {

cixio, 2024-10-01:
changed to `if (var == "xxxAUTODETECT_INFO") {`

bug is gone - and i dont know why 🤔

cixio, 2024-10-01:
To explain "removing the template processor"
Changing:
```
 server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(SPIFFS, "/index.html", String(), false, processor);
  });
```
to
```
  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(SPIFFS, "/index.html");
  });
```
Will make the page work on iOS 18 Safari (but there are some %variables% on the page, of course).

dl9rdz, 2024-10-02:
I can reproduce the problem with an iOS simulator. But the bug is not removed by changing the template processor. Rather, it seems that Safari on iOS does not handle HTTP chunked responses correctly. This is strange, as it is mandatory for a HTTP/1.1 browser to handle this.

dl9rdz, 2024-10-02:
fixed in dev20241002

cixio, 2024-10-02:
wow. I also found out that the problem has something to do with the chunked responses, after some attempts it was clear that there must be a problem with the ESPAsyncWebServer... after this I stopped debugging. Way to deep for me to solve it.

Btw: i like your comment regarding the RFC 🤭

Can confirm fix on real devices: iOS 18 and iPadOS 18.

closed 2024-10-02 by cixio