dl9rdz/rdz_ttgo_sonde #472: save parameter

Braccato, 2024-10-31:
Hi, with the new firmware 20241002 is it possible to save the parameter configuration?

dl9rdz, 2024-10-31:
Yes it should be

Braccato, 2024-10-31:
is there a guide?

dl9rdz, 2024-10-31:
What exactly do you want to achieve? Can you explain in more detail?

Braccato, 2024-10-31:
I wanted to save all the entered parameters to a file

dl9rdz, 2024-10-31:
Ah ok I see... 

You can do so on the command line with a python script in the repository. Something like (for saving, and for re-uploading to the ttgo)
`python scripts/ttgoconfig.py --ttgo=192.168.1.9 get config`
`python scripts/ttgoconfig.py --ttgo=192.168.1.9 put config`

You can also do the "save to file" simply with the web browser by accessing
http://<ip>/file/config.txt
But uploading via the web browser is only via http://<ip>/edit.html?file=config.txt and then copy/paste the saved file into it.

Braccato, 2024-10-31:
Grazie

closed 2024-12-31 by dl9rdz