dl9rdz/rdz_ttgo_sonde #476: API parameter/call for reset

jimjackii, 2024-11-08:
Hello,

Is there an API parameter/call to reset the device externally?

I would like to perform an automatic reset once a day because there are still problems with the device and it remains in AP mode after a few days (presumably restarted?).

The firmware is devel20240521, the WiFi mode is 1.

Regards, Steffen

dl9rdz, 2024-11-08:
You can do something like
`wget --post-data="reboot=1" http://[deviceip]/control.html`

ckuethe, 2025-01-09:
I've been thinking of adding a configurable feature to do this, either at a set time (perhaps 2245Z and 1045Z daily, so that it's reset a little bit before all the weather balloons go up), or a periodic timer (anyywhere from once a minute to once a year, as you deem appropriate)

jimjackii, 2025-01-10:
The configurable reset would be great.

I have the problem that sometimes the receiver is no longer accessible (WiFi starts in access mode) and I can't easily access the device to reset it manually. An internal automatic reset would be very helpful.