dl9rdz/rdz_ttgo_sonde #483: Connection failed with radiosondy and sondehub

Cofen86, 2024-12-06:
Good evening, I notice that with this latest version (dev20241002-C3) I am having problems with the connection to radiosondy and sondehub as per the attached screen.
[image]
The configured parameters are identical to those configured in dev1.
Does this happen to anyone else?
Thank you
IZ2MHJ

Braccato, 2024-12-07:
Yes, it keeps disconnecting for me too, I don't understand what it is

ckuethe, 2024-12-08:
I think, maybe, you can only have 1 connection at a time from a given source IP+Callsign. I was seeing the same thing; after removing the APRS2 host (just leave it blank), my connection has gone from flapping every 5 seconds to being stable for 7 minutes and counting.

dl9rdz, 2024-12-08:
Yes, with a single SSID you can have only one connection.

As soon as you log in to an aprs host with a second device, the first connection will be dropped. The usual solution is to append different SSIDs to the callsign (like DL9RDZ-1, DL9RDZ-2, etc.). But that is useful just if you have more than one device.

There is no point in connecting to radiosondy.info twice. Removing the second host should remove your problem.

Braccato, 2024-12-09:
Hi, even if I remove the host2 string it doesn't connect to the network, is it possible to have the complete configuration?

dl9rdz, 2024-12-09:
Ok, I see. Do you have IPv6 on your network?

There is one known issue in at least one tasmoto-espressif version (https://github.com/espressif/arduino-esp32/issues/8221) related to that. Older version of my software use older espressif version without IPv6 support, the current version uses tasmoto-espressif release 2024.08.10

Braccato, 2024-12-09:
Yes i have IPV6

ckuethe, 2024-12-09:
Might I suggest updating [Data-feed-configuration](https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Data-feed-configuration) to indicate that
* you should not have duplicate APRS-IS destinations (eg. both radiosondy.info ports), and
* you should (must?) use different SSIDs for each of your devices reporting to APRS-IS

dl9rdz, 2024-12-09:
Yes, done. I will also update the validation javascript code so you get a warning if you put the same server in both fields.

dl9rdz, 2024-12-10:
Please test with devel20241209. No big changes, but some debug info on DNS added.

[image]

Let me know your system shows in the new DNS line.

Braccato, 2024-12-10:
Ok I tested

dl9rdz, 2024-12-10:
Ok, what exactly is shown, the same IP address (192.168.1.1) as in my screenshot?
And below APRS is stuck in "DNS lookup"?

dl9rdz, 2024-12-10:
Alright, so DNS is an IPv6 address, that is good to know.
What IP does the TTGO have, I assume that is an IPv4?
And then accessing the IPv6 DNS server will fail... I added 8.8.8.8 as DNS3 (fallback) in the latest firmware, did this change anything regarding connecting to radiosondy? 

Deactivate GPS: depends on what you want
- Completely disable (no power, maybe to save battery): Modify the source code (pmu.cpp to not enable the GPS power on the power controller)...
- Just not use it: In the hardware config set "GPS RXD pin (-1 to disable)" to -1
- Not send your position to aprs/sondehub: set SondeHub location reporting and APRS location reporting to 0 (off) or 1 (fixed)

Cofen86, 2024-12-10:
Good evening, my rdz ttgo does not update to the new version.
clicking the "available dev2" button is displayed
"Doing update, wait until reboot" but the old version still remains....

Braccato, 2024-12-10:
turn off ttgo and do it again

Cofen86, 2024-12-10:
Already tried but nothing changes

dl9rdz, 2024-12-22:
Finally found a dual-stack Fritzbox AP for testing.
- The problem was that while the TTGO connected in IPv4 mode, the router sends RDNSS advertisement for the IPv6 DNS server, which the TTGO received and updated its DNS server (which it could not contact with IPv4)
- With an upgraded IDF version this problem seems to have gone.
- The additional workaround with setting DNS3 to 8.8.8.8 is still in place (but should not be needed anymore)
- If your TTGO is running a version with the DNS/IPv6 problem then also the OTA update is affected. As soon as the the TTGO receives an RDNSS advertisement it will stop being able to use DNS (and do OTA updates). There usually should be a time window between setting the correct DNS (with DHCP) and overwriting it with the wrong DNS (with the RDNSS broadcast). If not then you need to manually flash via USB...

- Another potential solution (not currently active) is to actually enable IPv6 on the TTGO. This seems to work nicely for outgoing connections (and then the IPv6 DNS server will also work), but not for incoming connections to the integrated web server. So no IPv6 only operation possible right now. I will leave this disabled for now.

dl9rdz, 2024-12-31:
Closing this issue as I think all problems are resolved now
- Unfortunately the "wrong" config with radiosondy.info being twice in the config was in the default config. This is fixed new
- The IPv6 issues should be resolved now as well... (IPv6 itself has not been activated yet, but so far noone has requested this feature anyway....)

closed 2024-12-31 by dl9rdz