dl9rdz/rdz_ttgo_sonde #490: time for prediction

mmadela, 2024-12-19:
I was wondering, why the prediction (e.g. approximate landing point) varies after a radiosonde stopped to move
When I am going to pick up one some hours after landing, the prediction might be completely wrong, if wind (weather condition) changed significantly.

This is due to current time taken to tawhiri API call
I would think that time from last received frame (at least for RS41) would be better option.

14ri004, 2024-12-19:
Hello, I had the same problem on a probe, I went hunting with the latest coordinates, the time to move on site, the position of the probe had moved in the meantime, I looked for the probe for 30 minutes without seeing it, before seeing that the installation point had changed by 800m, with the new coordinates up to date no worries to recover it, it's not the first time I have this joke

dl9rdz, 2024-12-19:
Right, this is a good suggestion.

But in any case, note that predictions in the past are rather limited in the API. The model data is basically just about the future – with a new model every 6h and a processing delay of maybe 3h you can go back in time something in between 3 to 9h, but not more. But sure, for immediately searching after landing, using the time of the last frame instead of the current time would work fine.

smithse, 2024-12-31:
Thought I would add that [https://radiosondy.info](url) does a pretty good job here (assuming that radiosonde in question has it's flight uploaded to the app) and will run a prediction at the time the sonde was last received. The predicted landing spot is then also added to the map. This might be the best option for cold hunts.

commit 17c36cff (2025-05-04): use rx time stamp for prediction (fix #490)
closed 2025-05-04 by dl9rdz