dl9rdz/rdz_ttgo_sonde #491: a password on the Config tab

14ri004, 2024-12-22:
Hello, would it be possible to put a password on the Config tab, this will allow me to share the Livemap on a forum, and without the possibility for visitors to be able to modify the configuration of my TTGO, they would just have the possibility of following live the probes received on my station, thank you.

14ri004, 2024-12-22:
it would be for 2 or 3 people max, it's just to see the decoding remotely on my ttgo, and prevent a person from being able to modify the parameters, I already use it like that on my forum, with 2 friends who can watch the live decoding of my station, it's been working perfectly for over a year and without any crashes, thank you for your work

dl9rdz, 2025-05-04:
Added this (without much testing) in the latest version (dev20250504).

Default is as used to be (all accessible without authentication), but you can edit user.txt

(http://<ip>/edit.html?file=user.txt)

First line is permissions for unauthenticated user. Default ",2," is all allowed, change to ",1,"
Second line is user account with permissions to modify: "<username>,2,<password>", default is "admin,2,admin".

closed 2025-05-04 by dl9rdz