dl9rdz/rdz_ttgo_sonde #503: ipv6

14ri004, 2025-01-03:
hello, I did not find the info, is TTGO RDZsonde compatible in ipv6??
thanks for the feedback

hr-ru, 2025-01-03:
Hi, IPv6 is not enabled by default.

- It would be a single line of code to enable IPv6 – but that is only for outgoing connections, so reporting e.g. to sondehub or an MQTT server would work with that (tested on a dual stack access point). 
- However, this is not sufficient to access the internal web server using IPv6 (i.e. on the dual stack access point, I could access the TTGO web server only using IPv4, not IPv6). 
- I have not tested anything with an IPv6 only access point yet.

I am currently looking at enabling TLS for the internal web server for some other use case, and this will cause some internal changes, so I am deferring looking closer into the IPv6 limitations until then.

14ri004, 2025-01-03:
ok thanks for your answer

closed 2025-01-05 by 14ri004