dl9rdz/rdz_ttgo_sonde #513: Additional UDP output

smithse, 2025-01-17:
I am toying with the development of an iOS 'SondeSeeker' app for the rdz_ttgo_sonde and wondering how much work would be involved to essentially clone the ChaseMapper UDP feature and have rdz send out a UDP packet with the same format as live.json?

I am currently polling the device via http to get the data and feel it might be better to have the data pushed out. Asking as the ChaseMapper format doesn't give all the data that's available via json, and it would be nice to have.

73,
Sean
VK6DI

smithse, 2025-01-17:
Update: I have looked into this and am working on it. Will create a PR once completed

dl9rdz, 2025-01-17:
You can also connect to TCP port 14570 where the JSON data will be pushed out periodically (once per second for sonde data), maybe this is an option instead of adding yet another way of communicating data?

smithse, 2025-01-17:
I did look at the TCP option with iOS and thought it might be a bridge too far in terms of the connection that has to be maintained. Anyway, I do have it sending the sonde2json data out on a UDP multicast address and am just trying to get the settings to display in the config page but I have pulled all my hair out now haha!