dl9rdz/rdz_ttgo_sonde #515: Rdzwx-go incompatibility problem

14ri004, 2025-01-19:
good evening, I have a small problem with the Rdzwx-go application, I have malfunctions on a tablet (Chinese) with Android 14, impossible to display my location, but it works well on Samsung S21 Android 14, and on another Tablet with Android 10, surely an incompatibility problem.

ckuethe, 2025-01-19:
do other apps like GPS test work? I have a cheap tablet too that claim to have a GPS but it seems not to. https://f-droid.org/packages/com.android.gpstest.osmdroid

14ri004, 2025-01-19:
good evening, indeed sometimes it doesn't show me a GPS??, with Google Maps, Whatsapp, sondehub and other applications, the position is OK, but with RDZwx-GO it doesn't find my position, and Mysondy Go doesn't start on it either, I reset the factory settings and that solved some localization problems, on my other Android 10 tablet (also Chinese) it's good everything works perfectly, no crash and position ok.

dl9rdz, 2025-01-19:
Hard to tell... 
- Please always state the version you are using (v1.0.7 or v1.1.0-candidate?). The 1.1.0-candidate has a newer android run time which may have changed something (to the better or the worse), so ideally test both versions
- If the Phone/tabled does not provide a location, then also the TTGO (if its one with GPS like the T-Beam) can provide position info to the app). 
- Also let me know if the apps asks you for location permissions? Maybe also check in the device settings what permissions the app has (it should ask you for location permission the first time you start it and then if you confirm remember this)

14ri004, 2025-01-19:
good evening hansi, I used v1.1.0 and now I put v1.0.7 but that does not solve the problem, the gps position works well on google maps, sondehub and whatsapp, only rdz-wx go does not give the position and does not communicate with the ttgo, the location is activated, authorization granted, it could perhaps be the IPv6 of the tablet which blocks access ??, the new tablet does not connect to the TTGO or the T-Beam, with 2 phones and 1 other tablet no worries everything works

closed 2025-02-15 by 14ri004