dl9rdz/rdz_ttgo_sonde #517: Compass view on OLED display

Nevvman18, 2025-01-25:
Hi!
Firstly, I'd like to congratulate and thank you for this project! It works flawlessly and helps much in sonde hunting!

I'd like to ask whether the compass view on OLED display is possible? After reading through the code and display configuration files ('screenXX.txt'), I saw that it is possible on some of the TFT displays. I tried changing the configuration for the OLED display, but I couldn't get it to work.
I read in the [issue 162](https://github.com/dl9rdz/rdz_ttgo_sonde/issues/162), that it is a planned feature (*Add compass display for OLED display (merge existing code from somewhere...)*).
I'd be awesome to have a display with compass, showing bearing to the sonde itself, and maybe also my course and sonde course (the 3 values accessible in configuration *relative to sonde:  GD, GI, GB: Distance, dIrection (absolute), relative Bearing*) and the altitude of the sonde.

Best,
Franek

Leo-PL, 2025-03-09:
This would be an absolutely killer feature for me. +1.

Nevvman18, 2025-03-09:
@dl9rdz 😁