dl9rdz/rdz_ttgo_sonde #518: WiFi password error

jimjackii, 2025-01-28:
**Describe the bug**
Whenever I set the WiFi password, a 7 (for me) appears at the very end after saving and restarting from client mode. This means that every further connection fails after a restart and I have to set the WiFi password again via the internal AP (with IP 192.168.4.1) and a single restart then works again.

**To Reproduce**
I have tested setting the WiFi password several times - the error remains the same.

**Version and specific configuration:**
rdzTTGOsonde version dev20250108
Autodetect info: Fingerprint 55 (TTGO T-Beam (V0.7), 0.9" OLED@21,22)

Regards, Steffen

dl9rdz, 2025-01-28:
There is one limitation I can confirm: Displaying the password in the form is limited to 31 characters. If you use a longer password, it will not show correctly when viewing the WiFi tab (and when you save again, the truncated shorter version will be stored). If you enter a longer password in the form, it should be correctly stored (and used later). 

You can always check the stored password by accessing `http://<ttgo ip>/edit.html?file=networks.txt`

Internal, there are basically no changes compared to previous version - except an upgrade of the idf environment that theoretically could have an impact. Then this "connection does not work" would be a completely independent problem.

jimjackii, 2025-01-28:
Ok, thank you very much.

Then I probably had the problem with saving the shortened password again. This never happened to me in the older version.

Otherwise I'll keep an eye on it.