dl9rdz/rdz_ttgo_sonde #523: SX1278 LNA Gain

73-de-LZ, 2025-02-14:
Looking trought the datasheet of SX1278 and RX_FSK.ino i found that LNA Gain is controlled by "RegLna 0x0C" and it is initialized once with that line in RX_FSK.ino:
`sx1278.setLNAGain(0);`
trought sx1278.cpp in the register should be written 0x20 or 0b00100000, which is perfectly fine for highest LNA gain.
Further this value will be changed by internal AGC, as it is enabled for some sondes.
But the interesting part of that register 0X0C is the last two bits, as described in the manual:
```
bits 1-0 --> LnaBoostHf High Frequency (RFI_HF) LNA current adjustment
00 --> Default LNA current
11 --> Boost on, 150% LNA current
```
And they are not changed by AGC. As i can't see this register to be changed anywhere else in the code i did this modification in RX_FSK.ino:
```
//sx1278.setLNAGain(0);
sx1278.writeRegister(0X0C, 0b00100011);
```
I did a test during sonde flights with two identical TTGO boards, fed by splitter from the same RF signal and i was unable to see undeniable difference of RSSI, but I have well filtered LNA, which i can't remove from the test setup easily. I did also try to measure the current, drawn by the boards, but since WiFi is working, it was hard to distinguish any difference also, so if somebody can do some real test and tell if there are some RX improvements, would be great to share the results here.

dl9rdz, 2026-11-10:
There is now a config option where you can turn on and off this feature.
If I have some spare time I will run two boards on a splitter on the same antenna in parallel and check if it makes any difference.

closed 2026-11-10 by dl9rdz

mmadela, 2026-11-11:
as far as I can understand the datasheet, the LNA boost works only for HF band, eg, not radiosonde frequencies

dl9rdz, 2026-11-11:
Indeed. I also just checked. Seems like this is only for the "high band" LNA (i.e. for 800 MHz operation) not the "low band" LNA used for 137MHz/400MHz, so very likely useless...