dl9rdz/rdz_ttgo_sonde #524: probe voltage on TTGO

14ri004, 2025-02-18:
Hello, and is it possible or not ?? to display the voltage of the probe on TTGO, it seems useful to me to know the state of the battery and its lifespan, in order to avoid unnecessary travel, the M20 have an unfortunate tendency (often) to cut off emissions well before the ground, which becomes difficult to anticipate the departure for the hunt, especially at more than 50km, this would be a valuable help as with the M10, thank you

dl9rdz, 2025-02-18:
It is possible. Please test and report if I got it right (no M20 for testing here....)

14ri004, 2025-02-18:
on my T-BEAM, the voltage of the Probe is well indicated on the M10 and the RS41 with this (7.8,20,6=mbv), but with the M20 it does not work, and the display remains black

dl9rdz, 2025-02-18:
With devel20250218?

14ri004, 2025-02-18:
rdzTTGOsonde version dev20250210

dl9rdz, 2025-02-18:
I see. Well, I should have written a longer message then...

What I wanted to tell you is that I just added this functionality in the most recent version (dev20240218). Could you test that version and tell me if it is doing what it should do?

14ri004, 2025-02-18:
I will test it tomorrow to see, and keep you informed of the results.

14ri004, 2025-02-18:
perfect it works, thank you for your work

[image]

closed 2025-02-19 by 14ri004