dl9rdz/rdz_ttgo_sonde #525: Login - how to ?

AdamM68, 2025-02-19:
Is there an functional option to log into the page ?
There is a button in lower left corner [login] - but it looks like it isn't working... it shows bad credentials 
How can I make my page visible only after logging in ?

dl9rdz, 2025-02-19:
Oh this accidentally made it into the repo already, it still under development.
I want to protect access to configurations, and that in a reasonable way (without http).
You can login (with password "1234") but the configuration settings how things should be organized (all as it was vs read/only without authentication but config changes need auth vs. no access without login) still is not included [will be added 
soon-ish]

dl9rdz, 2025-05-04:
See #491 - now available in the latest dev2 version.

Default user is admin password admin, change in user.txt

closed 2025-05-04 by dl9rdz