dl9rdz/rdz_ttgo_sonde #526: bad RS41 reception

14ri004, 2025-02-22:
good evening, small problem on RS41, that makes 5 RS41 probes that are not decoded, nobody has sent frames, invisible on sondehub, I receive the frequency, the RS41 type, and nothing else (see screenshot)

T-BEAM v1.1 with rdzTTGOsonde version dev20250218

[image]

dl9rdz, 2025-02-22:
Could this be a military RS41SGM with encryption? If nobody, also not with other software, decoded these, I strongly suspect that this is very likely.

Those will send the serial number in clear text but position encrypted.

14ri004, 2025-02-22:
YES that's what I thought too military coding, thanks

vg1320, 2025-02-22:
Hi,

yes the V0541501 was detected as a RS41 SGM :

Timestamp: 2025-02-22T17:37:13.714098
Sonde flight detected in area:

ENCRYPTED RADIOSONDE DETECTED!
Snr:       6.3 
Batt:      2.90 V
Serial:  V0541501
Type:      RS41-SGM
Frequency: 402.7000 MHz
Time Detected: 2025-02-22T16:37:12.342615+00:00Z

There were 7 RS41 SGM detected from stations in France today

closed 2025-02-24 by 14ri004