dl9rdz/rdz_ttgo_sonde #528: SD card

pavolgaj, 2025-03-05:
Status of SD card is "SD card init failed" after adding the SD card. I use TTGO LoRa32 with latest dev firmware. SD card fully works in PC. It's formatted as FAT.
I already tested it before in some TTGO and myloc file was created there. Maybe it was the same TTGO as I use now...

dl9rdz, 2025-03-05:
Not easy to tell. Maybe you could share your full config?

- No changes to functionality, and the SD card is working with the latest dev firmware on my testing TTGO.
- Double-check your pin configuration (make sure you have all your pins configured correctly, and also make sure you don't have any SD card pins configured elsewhere, e.g. as buttons)
- It needs to be the plain and simple FAT file system, specifically exFAT is not supported.

pavolgaj, 2025-03-05:
I don't change anything in config (using default values). I don't have anything connected else to TTGO. Yes, file system is standard FAT.

dl9rdz, 2025-03-05:
Maybe one more thing: SD cards needs to be inserted before powering on the TTGO. The TTGO will attempt initialization only once on power up. If you insert the SD card later, it will just show the state (init failed) on power up.

pavolgaj, 2025-03-05:
I have connected a battery to TTGO. Is turning off/on by build-in switch enough? Or do I need to unplug the battery? It could be a bit problematic due to the plastic box...

dl9rdz, 2025-03-05:
Turning off/on is enough

pavolgaj, 2025-03-05:
Unfortunately, it doesn't work for me. I also tried to plug USB cable and powered it only by cable (with switch off). But, it didn't help me:(

d2f2, 2025-03-20:
> I tried to format the SD card to FAT16 (it was FAT32). Not help... Maybe, SD slot on TTGO is broken. I'll have to try it with a different board. But, I only have access to this one:(

Please share your hardware config. It seems that you did not provide correct configuration for SD card pins.
What is your board? LILYGO? Please share hardware version info.

d2f2, 2025-03-20:
> Not easy to tell. Maybe you could share your full config?
>
> * No changes to functionality, and the SD card is working with the latest dev firmware on my testing TTGO.
> * Double-check your pin [...]

Have you any plans to implement SD card format funtion? Or file remove funtion?
I have my pcb above roof, at the side of antenna, and have to format SD card soon. No space left in few more days.... ;)
Or can i do this nów, but did not know how? ;)

pavolgaj, 2025-03-20:
I use this TTGO board: https://lilygo.cc/products/lora3 without any additional connected devices. I didn't change any pin configuration in the firmware (I use default ones).

d2f2, 2025-03-21:
> I use this TTGO board: https://lilygo.cc/products/lora3 without any additional connected devices. I didn't change any pin configuration in the firmware (I use default ones).

So show those default ones.

pavolgaj, 2025-04-12:
Sorry for late answer - I didn't have access to TTGO.

The config for SD card is:
CS - 13
MISO - 2
MOSI - 15
CLK - 14

dl9rdz, 2026-11-10:
I think this can be closed :-)
Formatting via web interface is possible by now
Deleting files as well

closed 2026-11-10 by dl9rdz