dl9rdz/rdz_ttgo_sonde #530: Feature Request, easy enhancement?

va24531, 2025-03-31:
IMHO, it would be nice to see the little more of a expanded mqtt status report.
For instance, If a topic of 'action' for scanning, receiving or single ( as in what frequency it was listening to, but not receiving packets) would be nice. 

I personally have my board (TTGO Lora32 v2.1_1.6) enclosed at the top of a 48' pushup pole, power being supplied via usb cable from a small solar panel mounted at ground level. Being able to listen for the battery voltage via mqtt instead of having to scrape/ask for it from the home page would be welcomed!)

Mainly, I'd love to see at least the receiver activity being pushed out via mqtt.

hr-ru, 2025-03-31:
Did you look at the "MQTT message selection bitfield" config field (in recent dev2 versions)?

If activated you will get MQTT messages to the "qrg" topic with what frequency it is listening on, and the "pmu" topic with battery information. Sounds like exactly what you want to have?

va24531, 2025-04-01:
Yeah, when I saw that option, I was really hopeful, but for some reason, when I am only able to set mode 1 and mode 2. If I go higher, the board resets and looses all it's settings. I'm looking into that, and maybe it's a bad combination of settings or something.

board: Autodetect info: Fingerprint 31 (TTGO LORA32 v2.1_1.6 (0.9" OLED@21,22))

hr-ru, 2025-04-01:
Ah ok I have never tested that part of the code in depth myself, this is code that was contributed....

ckuethe, 2025-04-02:
it worked on my TTGO devices with all the different bit masks. I haven't reinstalled them lately as they're doing what I need them to do.

One of my remote nodes just reported `{"I_Batt": 0.0, "V_Batt": 3.335, "I_Vbus": 0.0, "V_Vbus": 3.363, "T_sys": 12.0}`. According to the uptime topic it's running a build from December. `{"uptime": 112.8, "user": "radiosonde", "time": 2025-04-02 01:56:15, "rxlat": 35, "rxlon": -120, "SW": "rdzTTGOsonde", "VER": "dev20241225"}`