dl9rdz/rdz_ttgo_sonde #533: additional functions

14ri004, 2025-04-20:
Hello Hansi, I have 2 questions for you, 1) on the next Devel would it be possible to have an access code to block the TTGO parameters, in order to prevent anyone from changing the parameters, this would be really useful to me, 2) can you (if possible obviously) add the ST7796S screen drivers for a 4" TFT screen, thank you for your work

dl9rdz, 2025-04-20:
1) This is on my roadmap, the "login" mechanism (accidently) made it already in the latest devel (it is just not doing anything usefyl yet), but the plan is to combine this with allowing certain actions only after login.

2). I don't have such display to test, but maybe you can do that? Maybe for a quick test, just change this line in the source code: https://github.com/dl9rdz/rdz_ttgo_sonde/blob/dev2/RX_FSK/src/Display.cpp#L532
replace Arduino_ST7789 by Arduino_ST7796 (and then select ST7789 in the config).  Does this work?
If yes then I can add this as config option. Of course one would need to make a different screens.txt, and adjusting the spectrum for the different resolution would be additional work.

14ri004, 2025-04-20:
Regarding the settings lock, this could be interesting. I hope you find it.

I'd like to try a 4" screen, but since the drivers aren't included in ttgo, I'm waiting for confirmation before purchasing this screen. If you can add it or replace it with another one, I'll order the 4" TFT. I'll keep you posted on the results. The 3.2" already works perfectly. I've been using it on a T-Beam for four years. I wanted to try something a bit larger. Let me know if you can add the 4" drivers. Thanks.

kmp56, 2025-04-21:
@dl9rdz  This solution works without any problems with the ILI7796 4" display. To fully use the 480x320 size, you must of course edit the appropriate sceensX.txt file - it is best to create a new file with a new name...

closed 2025-05-07 by 14ri004