dl9rdz/rdz_ttgo_sonde #536: small problem the GPS point and coordinates are no longer displayed on livemap.html

14ri004, 2025-05-05:
Hello Hansi, the code works very well to block access to the T-beam settings, but there is a small problem: the GPS point and the coordinates are no longer displayed on livemap.html. On the T-Beam screen, the coordinates are present, but not visible on the livemap page. There is a blockage somewhere. Thank you.

[image]

[image]

dl9rdz, 2025-05-06:
This is strange, I cannot reproduce this here. 

The GPS position of the TTGO (missing icon where your red arrow points to) does not show on the map if (a) there is not GPS reception of the T-Beam and (b) no fixed position is entered in the config.  Then of course the TTGO does not know its own position and does not show it on the map.   This is the same behaviour as in previous versions.

The position of a received sonde signal shows on the map in any case.

14ri004, 2025-05-06:
Hello Hansi, I may have explained myself badly, my T-BEAM is in the car, with the dev20250401-C3-full.bin version I had my GPS position displayed on the Livemap, but after putting the dev20250504-C3-full.bin my position is no longer visible, on the T-BEAM I have the coordinates of my position, but they are no longer visible on the map, so I no longer have the blue dot, nor the distance between me and the probe, impossible to locate myself on the map, I will try with another T-BEAM in version dev20250401-C3-full.bin to be sure of my problem, I will keep you informed of the rest, thank you

dl9rdz, 2025-05-06:
Ok, so this should be independent of the login / password feature then I assume?

Can you tell me
(a) do you have your "home" position set (i.e. in the config "receiver fixed lat/lon")?
(b) Can you tell me what the main status page ("Home" tab) on the web interfaces shows in the "GPS" part?
(c) Tell me what the live.json on your TTGO looks like (http://<ttgoip>/live.json)

14ri004, 2025-05-06:
I just reprogrammed another T-BEAM with version dev20250401, and I'm doing some tests, I'll tell you in 30 minutes to do the test between the 2 T-BEAMs, but before updating the one in the car to work perfectly, I use the GPS I don't have any coordinates in the config, it's been working like that for 3 years, apart from having done the update nothing else has changed

14ri004, 2025-05-06:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

14ri004, 2025-05-06:
rdzTTGOsonde version dev20250506

[image]

Thanks Hansi, great work, with the 506 it's ok I found the position on livemap, thank you very much

14ri004, 2025-05-06:
I have another question, for the admin password, how many characters can it have at most? I have a friend who blocked his T-BEAM, he must have made a mistake, impossible to enter the T-BEAM (lol) his password is invalid.

dl9rdz, 2025-05-06:
There is a limit of 128 characters for one full line in the user.txt file. 

I just tested with a 116 character password and this still works fine...

14ri004, 2025-05-06:
ok merci

closed 2025-05-06 by 14ri004