dl9rdz/rdz_ttgo_sonde #538: Upload to sondehub not working

DL4OCE, 2025-05-06:
While I can see successful uploads to the radiosondy database, I only get an error message from sondehub:

Sondehub	State: Connected: Sending data. Last upload start: 4 s ago
Last reply: HTTP/1.1 202 Accepted {"message": "some or all payloads could not be processed", "errors": [{"error_message": "DFM radiosonde above 1000 and not enough data to perform z-check. Not adding DB to protect against double frequency usage.", "payload": {"software_name": "rdzTTGOsonde", "software_version": "dev20250504", "uploader_callsign": "DL4OCE", "time_received": "2025-05-06T13:12:42.000Z", "manufacturer": "Graw", "serial": "23053868", "datetime": "2025-05-06T13:12:41.000Z", "lat": 52.XXXXX, "lon": 9.XXXXX, "**alt": 14691.7998**, "frequency": 403.29, "vel_h": 4.31, "vel_v": 4.07, "heading": 89.46, "rssi": -113.0, "frame": 1430572361, "type": "DFM"

Altitude is 81 meters, in reality, and also set in the GUI.

**To Reproduce**
Steps to reproduce the behavior:
Open https://sondehub.org/, select an active sonde and never see any entries in the callsign list, neither will a line be drawn to my ground station. Ground station is visible on the map.

**Version and specific configuration:**
 - What firmware version are you using? binary devel or master image? compiled code on your own, possibly with modification?
dev20250504, installed using the GUI
  
 - Any specific settings
 - Hardware you are using? (ESP32 board? Display? Additional hardware connected to ESP32?)
Autodetect info: Fingerprint 31 (TTGO LORA32 v2.1_1.6 (0.9" OLED@21,22))

DL4OCE, 2025-05-06:
In the device's web GUI, it shows received sondes.

dl9rdz, 2025-05-06:
This seems to be related to a merge request that I indeed should merge.

The problem is that my software sends updates to SondeHub every 15 seconds. For DFM, SondeHub requires that one single upload contains at least 10 different. The recommend to collect frames for at least 20 seconds (https://github.com/projecthorus/sondehub-infra/wiki/DFM-radiosonde-above-1000-and-not-enough-data-to-perform-z-check).

In perfect conditions, you should decode > 10 frames within 15 seconds, but in non-ideal conditions, if you decode only 9 frames or less within the 15 seconds, Sondehub will ignore your upload with the error message that you are seeing. 
I will make some change in the next version, making this longer (or adjustable in the config).

DL4OCE, 2025-05-06:
So, I *was* getting the error message wrong, as I thought that my altitude was reported with 14691 (whatever unit).

Looking forward to receiving git's update about the merged version :-)

ae5au, 2025-05-16:
> This seems to be related to a merge request that I indeed should merge.

@dl9rdz if you are referring to my PR #471 and interested in merging it, I'm happy to bring it current and test this weekend.

dl9rdz, 2025-12-16:
This is still on my todo list. I wanted to test this myself but was unable to reproduce. Probably need to have some specific unstable network connections for this to happen. I think the merge request is a good idea in any case.

Let me merge this in the next release, but without much testing done by myself.