dl9rdz/rdz_ttgo_sonde #539: RDZ will not connect to an open wifi network, specifically to my hAP Mikrotik router.

9a3zi, 2025-05-11:
any ideas? 73.

dl9rdz, 2025-05-11:
Hard to tell. The closest I have at hand right now is a Mikrotik map2.

Mikrotik default configuration, renamed Wifi net to "mikrotik"
Configured Wifi on TTGO with name "mikrotik" as name and empty password

Works for me.
```
RX_FSK: [WiFi-event] event: 101
WiFi interface ready
RX_FSK: [WiFi-event] event: 110
WiFi client started
RX_FSK: [WiFi-event] event: 102
Completed scan for access points
RX_FSK: Network DIRECT-FI-EPSON-ET-2850 Series: RSSI -58, MAC E2:BB:9E:0F:11:87, enc: WPA2_PSK
RX_FSK: Network mikrotik: RSSI -60, MAC 4C:5E:0C:D6:C9:2F, enc: Open
RX_FSK: Match for mikrotik at 1
RX_FSK: Match found at scan entry 1, config network 1
RX_FSK: Network Dinosauro: RSSI -79, MAC A4:17:8B:7F:23:88, enc: WPA2_PSK
RX_FSK: Match for Dinosauro at 2
Connecting to: mikrotik with password 
RX_FSK: [WiFi-event] event: 112
Connected to access point
.........RX_FSK: [WiFi-event] event: 115
Obtained IP address: 192.168.88.253
.
WiFi connected
IP address:
192.168.88.253
```

You could provide serial port log from the TTGO, maybe this shows something that helps identifying the problem.
But basically this should work.

(What is not supported is a open WiFi AP that then uses a captive portal for authentication...)

9a3zi, 2025-05-13:
thanks for the quick reply, solved thanks to you.
73

closed 2026-11-13 by dl9rdz