dl9rdz/rdz_ttgo_sonde #540: SD card format

d2f2, 2025-05-12:
Hello
Is this possible to format SD card via web?
Seems it be full card of my high roof Mount rdz. How to format remote'ly?

dl9rdz, 2025-05-12:
Currently there is no such feature. Should be easy to add though.  (If added to the next version I assume you could then OTA update your device remotely...)

d2f2, 2025-05-18:
Will be great have Such feature. Looking forward for it. Thank you for your great job!

Ota is great

kmp56, 2025-05-24:
@dl9rdz 
Ubuntu 24.04 + Visual Studio Code
Unfortunately, after "Project has been successfully updated!" error while compiling .ino:

`/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino: In function 'void execRinexUpdate()':
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino:2980:4: error: 'geteph' was not declared in this scope; did you mean 'getopt'?
 2980 |    geteph();
      |    ^~~~~~
      |    getopt
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino:2982:8: error: 'ephstate' was not declared in this scope
 2982 |    if (ephstate == EPH_PENDING) ephstate = EPH_ERROR;
      |        ^~~~~~~~
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino:2982:20: error: 'EPH_PENDING' was not declared in this scope; did you mean 'ETS_PENDING'?
 2982 |    if (ephstate == EPH_PENDING) ephstate = EPH_ERROR;
      |                    ^~~~~~~~~~~
      |                    ETS_PENDING
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino:2982:44: error: 'EPH_ERROR' was not declared in this scope; did you mean 'RX_ERROR'?
 2982 |    if (ephstate == EPH_PENDING) ephstate = EPH_ERROR;
      |                                            ^~~~~~~~~
      |                                            RX_ERROR
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino:2983:4: error: 'get_eph' was not declared in this scope
 2983 |    get_eph("/brdc");
      |    ^~~~~~~
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino: In function 'void execFormatSD()':
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino:2997:4: error: 'connSDCard' was not declared in this scope
 2997 |    connSDCard.format();
      |    ^~~~~~~~~~
Compiling .pio/build/ttgo-lora32/lib5f6/Network/NetworkClient.cpp.o
Compiling .pio/build/ttgo-lora32/lib5f6/Network/NetworkEvents.cpp.o
Compiling .pio/build/ttgo-lora32/lib5f6/Network/NetworkInterface.cpp.o
Compiling .pio/build/ttgo-lora32/lib5f6/Network/NetworkManager.cpp.o
Compiling .pio/build/ttgo-lora32/lib5f6/Network/NetworkServer.cpp.o
Compiling .pio/build/ttgo-lora32/lib5f6/Network/NetworkUdp.cpp.o
Compiling .pio/build/ttgo-lora32/lib9fd/AsyncTCP@src-2154f270fe2afd37e50c673663dbc05d/AsyncTCP.cpp.o
Compiling .pio/build/ttgo-lora32/lib1b1/FS/FS.cpp.o
Compiling .pio/build/ttgo-lora32/lib1b1/FS/vfs_api.cpp.o
Compiling .pio/build/ttgo-lora32/libc12/WiFi/AP.cpp.o
*** [.pio/build/ttgo-lora32/src/RX_FSK.ino.cpp.o] Error 1`

Compilation ver. 2025-05-16 trouble-free...

dl9rdz, 2025-05-24:
Indeed it does compile only if RS92 feature is enabled. I have fixed that, now it should also compile again with RS92 feature turned off.

kmp56, 2025-05-24:
@dl9rdz 
still with error:

`Compiling .pio/build/ttgo-lora32/lib410/ESP Async WebServer/WebServer.cpp.o
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino: In function 'void execFormatSD()':
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino:3008:4: error: 'connSDCard' was not declared in this scope
 3008 |    connSDCard.format();
      |    ^~~~~~~~~~
Archiving .pio/build/ttgo-lora32/libc70/libU8g2.a
Indexing .pio/build/ttgo-lora32/libc70/libU8g2.a
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/Arduino_DataBus.cpp.o
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/Arduino_G.cpp.o
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/Arduino_GFX.cpp.o
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/Arduino_GFX_Library.cpp.o
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/Arduino_TFT.cpp.o
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/Arduino_TFT_18bit.cpp.o
Archiving .pio/build/ttgo-lora32/lib1b1/libFS.a
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/canvas/Arduino_Canvas.cpp.o
Indexing .pio/build/ttgo-lora32/lib1b1/libFS.a
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/canvas/Arduino_Canvas_3bit.cpp.o
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/canvas/Arduino_Canvas_Indexed.cpp.o
Compiling .pio/build/ttgo-lora32/lib0d8/GFX Library for Arduino/canvas/Arduino_Canvas_Mono.cpp.o
*** [.pio/build/ttgo-lora32/src/RX_FSK.ino.cpp.o] Error 1`

dl9rdz, 2025-05-24:
Fixed as well.

kmp56, 2025-05-24:
> Fixed as well.

Still error: 

`/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino: In function 'void execFormatSD()':
/home/kmp/Arduino/-2025_RDZ_GOTOWE/2025-05-23/TFT/3.2inch_TTGO-LoRa32_v2.0_2.1_2025-05-23-IGORR/RX_FSK/RX_FSK.ino:3008:4: error: 'connSDCard' was not declared in this scope
 3008 |    connSDCard.format();
      |    ^~~~~~~~~~
*** [.pio/build/ttgo-lora32/src/RX_FSK.ino.cpp.o] Error 1`

kmp56, 2025-05-24:
@dl9rdz 
Indeed, now the latest github version compiles with the SD card enabled and disabled, but I don't have an SD card today and I might have one tomorrow... And unfortunately this compilation with the SD enabled interferes with the LCD screen - black screen.
BTW. Great job! Thanks for the support!

dl9rdz, 2025-05-24:
> all SD card settings: 6x 0 (zero)

If you want to disable SD card in config, set them to -1 (at least clk and cs)

kmp56, 2025-05-24:
@dl9rdz 
I don't really plan on using an SD card, so that's not a problem for me. Thanks for your help and explanations!

closed 2026-11-10 by dl9rdz