dl9rdz/rdz_ttgo_sonde #541: 4" TFT screen on my T-Beam

14ri004, 2025-05-16:
Hello Hansi, I just finished mounting the 4" TFT screen on my T-Beam, can you add the driver in the devel, honestly it gives a good 4" screen

dl9rdz, 2025-05-17:
ok (untested driver can be selected with display type 6.

14ri004, 2025-05-17:
Hello Hansi, that's perfect, thank you and good job,

[image]

14ri004, 2025-05-17:
[image]

kmp56, 2025-05-17:
unfortunately, the spectrum screen remains unscaled 240x320...

kmp56, 2025-05-17:
@14ri004
My screen file for 4": 

##########
# Scan display for 4" TFT dispaly [ST7796]
@Scan.TFT.kmp
timer=-1,0,0
key1action=D,#,#,#
key2action=#,#,F,#
timeaction=#,D,+
scale=10,10
color=00FFFF
##fonts=2
1.5,18=XSonde Scanner
fonts=1
color=FFD700
7,1=xLaunch place
color=DCDCDC
7,30=S
#Row 1
color=FFD700
10,1=xModel
color=DCDCDC
10,30=T
color=FFD700
13,1=xFrequency
color=DCDCDC
13,30=F
13,40=xMHz
19,1=x----------------------------------------------
#Footer
color=FFFF00
21,1=xYour current
22.7,3.3=xposition:
color=DCDCDC
22,20=GA
22,30=GO
color=C0C0C0
###fonts=1
29,1=xIP for configuration: 
color=00FF00
fonts=0
29,30=N

############
@Decode/General.TFT.kmp
timer=-1,-1,N
key1action=+,0,#,#
key2action=>,#,F,#
timeaction=#,#,0
scale=10,10
color=FFFFFF
fonts=1
0,35=xLaunch place
color=FFFF00
##fonts=1
2,35=s
##fonts=0,2
color=FFFFFF
0,1=t
0,8=xSerial
color=FFFF00
2,1=Is
color=FFFFFF
0,17=xFrequency MHz
color=FFFF00
2,17=F
color=808080
4,1=x----------------------------------------------
#Row 2
color=9EB5E3
fonts=0
7,1=xLatitude
color=FFFFFF
fonts=1
9,1=L
color=9EB5E3
fonts=0
7,17=xLongitude
color=FFFFFF
fonts=1
9,17=O
color=9EB5E3
fonts=0
7,35=xAltitude
color=FFFFFF
fonts=1
9,35=A
#Row 3
color=9EB5E3
fonts=0
12,1=xHoriz. Speed
color=FF00FF
fonts=1
14,1=Hkkm/h
color=9EB5E3
fonts=0
12,17=xVert. Speed
color=FF00FF
fonts=1
14,17=Vm/s
color=9EB5E3
fonts=0
12,35=xTemperature
color=FFFA02
fonts=1
14,35=Mt 
13,41=xc
#Row 4
color=9EB5E3
fonts=0
17,1=xSignal
color=00FF00
fonts=1
19,1=Q4
color=9EB5E3
fonts=0
17,17=xRSSI
color=00FF00
fonts=1
19,17=R
19,22=xdBm
color=9EB5E3
fonts=0
17,35=xPressure
color=FFFA02
fonts=1
19,35=Mp
19,43=xhPa
#Row 5
color=9EB5E3
fonts=0
22,1=xBearing
color=00FFFF
fonts=1
24,1=GB
color=9EB5E3
fonts=0
22,17=xDistance
color=00FFFF
fonts=1
24,17=GD
color=9EB5E3
fonts=0
22,35=xHumidity
color=FFFA02
fonts=1
24,35=Mh
24,40=x%
color=808080
26.5,1=x--------------------------------------------------
color=FFFF00
27.7,1=xYour current
29.5,3.3=xposition:
color=DCDCDC
29.5,20=GA
29.5,30=GO

#################
@Compass/General.TFT.kmp
timer=-1,-1,N
key1action=+,#,#,#
key2action=#,0,F,#
timeaction=#,#,#
scale=10,10
color=FFFFFF
fonts=1
0,35=xLaunch place
color=FFFF00
##fonts=1
2,35=s
##fonts=0,2
color=FFFFFF
0,1=t
0,8=xSerial
color=FFFF00
2,1=Is
color=FFFFFF
0,17=xFrequency MHz
color=FFFF00
2,17=F
color=808080
4,1=x----------------------------------------------
#Row 2
color=9EB5E3
fonts=0
7,1=xLatitude
color=FFFFFF
fonts=1
9,1=L
color=9EB5E3
fonts=0
7,17=xLongitude
color=FFFFFF
fonts=1
9,17=O
color=9EB5E3
fonts=0
7,35=xAltitude
color=FFFFFF
fonts=1
9,35=A
fonts=0
#color=FF4500
## color=9EB5E3
## 8.5,4.6=xBearing
## color=00FFFF
## 9.5,5.3=GB
color=9EB5E3
15,35=xDistance
color=00FFFF
17,35=GD
fonts=0
###color=00FFFF
## 11.8,3=g0NCS,31,ffff00,000044,3,33ff33,4,FF4500
color=FFFFFF
11.1,7=g0CCS,60,ffff00,000044,7,ffff00,5,FFFF00
color=808080
25,1=x----------------------------------------------
fonts=1
color=9EB5E3
27,5=xSignal
color=00FF00
30,3.5=Q4
color=9EB5E3
27,32=xRSSI
color=00FF00
29.6,30=R
29.6,35=xdBm
#######
#######

14ri004, 2025-05-17:
Good evening, it's good I made the file for the 4" screen it's already much better now,

closed 2025-05-19 by 14ri004