dl9rdz/rdz_ttgo_sonde #552: 4.0"  TFT screen ST7796 test

SQ3H, 2025-07-08:
software: dev20250523-C3-full
hardware: t-beam v1.2, noname china 4.0" TFT SPI 480x320px v1.1

 - the software does not automatically recognize the screen type
 - screen type is not shown in the selection list in the hardware configuration menu
 - after manually entering the value "6" - it works

I didn't notice any other errors during use

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

AdamM68, 2025-07-09:
on picture 4 I can see: Display type = 5 .  Should it be 6 as You described above ?

dl9rdz, 2025-07-09:
On the small TTGO (LoRa32 V2.1_1.6, not a T-Beam):
The following wiring and configuration works (but others work as well):
[image]

GPIO15 - SDA [sometimes labelled SDI or MOSI on display]
GPIO14 - CLK [sometimes labelled SCL or SCK on display]
GPIO21 - RS [sometimes labelled D/C on display]
GPIO4 - CS

other pins on display:  GND=GND, VCC=3V3, LED=3V3 (depends on the TFT circuit. Often there is a transistor pulling the backlight LED cathode to GND if input is high. often there is an on-board pull up, so you can leave this pin open(unconnected). Pull it LOW (GND) to disable the backlight LED). 
(I connected RST from display to GPIO23 on my TTGO LoRa v2.1, which is the SX1278 reset. so the display is reset on startup by the SX1278 reset, and I put -1 in the config.  You can of course also use a different unused pin and put that (instead of the -1) in the config).

SQ3H, 2025-07-09:
Yes, it works great on TTGO, but it needs manual settings

[image]

[image]

[image]