dl9rdz/rdz_ttgo_sonde #560: T-BEAM V1.1 0.9-Zoll OLED with main0.98-C3-full.bin doesn't work

TFS-GH, 2025-08-12:
I updated the software on my T-Beam V1.1 with a 0.9-inch OLED display as follows: 3.3V, GND, 4, 13, and from master_v0.9.5-B17-full.bin to main0.98-C3-full.bin.
Since then, the OLED display has stopped working.
If I install the old master_v0.9.5-B17-full.bin, the OLED display works again.
Screen config = 1
Display type = 0

What could be the problem?

dl9rdz, 2025-08-12:
- Check how the board is auto detected. I think you have an unusual OLED connection (usually the OLED is on pins 21, 22 on the T-Beam 1.0/1.1), so you have to configure pins manually
- One difference to the old software is that in case the board is detected (or mis-detected in your case) as the msall V2.1_1.6 board, the SD card is activated by default, and that may overlap with your OLED configuration. Make sure to disable the SD card (set all pins to -1).

TFS-GH, 2025-08-12:
Hallo,

danke für die schnelle Antwort.
Ich mache mal alles bei der SDCard auf -1
Autodetect info: Fingerprint 55 (TTGO T-Beam (V0.7), 0.9" ***@***.***,22)
2025-08-12T21:47:36, Uptime: 0d 00:00:32, WiFi uptime: 0d 00:00:20, Batt: 0.00V
rdzwxGO app: not connected
RS92 RINEX eph state: Disabled (no RS92 in QRG list or Wifi mode not 3) 
ich dachte die PINS  21 22 sollte man wegen dem Powercontroller ab T-BEAM 1.0 nicht mehr nehmen.
Bei dem 2 Zoll Display geht auch alles mit den PIN`s nur das OLED will nicht 
 
 
--
Mit freundlichen Grüßen
Thomas Fröhlich
 
Mail   ***@***.***
Web   : http://www.tfs-web.de

====================== Original Nachrichtentext ===================
am Dienstag, 12. August 2025 um 23:39 schrieben Sie:

dl9rdz left a comment (dl9rdz/rdz_ttgo_sonde#560)
* Check how the board is auto detected. I think you have an unusual OLED connection (usually the OLED is on pins 21, 22 on the T-Beam 1.0/1.1), so you have to configure pins manually
* One difference to the old software is that in case the board is detected (or mis-detected in your case) as the msall V2.1_1.6 board, the SD card is activated by default, and that may overlap with your OLED configuration. Make sure to disable the SD card (set all pins to -1).
—

================= Ende des Original Nachrichtentextes =============
------------12A0F41E7321ED663
Content-Type: text/html; charset=utf-8
Content-Transfer-Encoding: quoted-printable

<html><head>
        <style type=3D"text/css" title=3D"rt_noDelete">
    blockquote.rt {
    margin: 0 0 15px;
    border-left: 4px solid #81c784;
    padding: 0 0 0 12px;
    display: block;
    }
    p { margin: 0 0 0 0 }
.email-signature {font-family:"Arial"; font-size: 8pt; font-style: italic; =
font-weight: normal; text-decoration: none; }
    </style></head><body>
<span style=3D" font-family:'Verdana'; font-size: 10pt;">Hallo,<br/>
<br/></span><p><span style=3D" font-family:'Verdana'; font-size: 10pt;">
</span><span style=3D" font-family:'Verdana'; font-size: 10pt;"></span></p>=
<p>danke f=C3=BCr die schnelle Antwort.</p><p>Ich mache mal alles bei der S=
DCard auf -1<br/></p><table class=3D"status-table" style=3D"border: 1px sol=
id black; border-collapse: collapse; background-color: rgb(221, 221, 221); =
width: 1366px; margin-top: 20px; color: rgb(0, 0, 0); font-family: Arial, s=
ans-serif; font-size: medium; font-style: normal; font-variant-ligatures: n=
ormal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal;=
 orphans: 2; text-align: left; text-transform: none; widows: 2; word-spacin=
g: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoratio=
n-thickness: initial; text-decoration-style: initial; text-decoration-color=
: initial;"><tbody id=3D"status-body"><tr><td style=3D"border: 1px solid rg=
b(204, 204, 204); border-collapse: collapse; background-color: rgb(221, 221=
, 221); padding: 10px; text-align: left; vertical-align: top;"><span style=
=3D"color: rgb(0, 0, 0); font-family: Arial, sans-serif; font-size: medium;=
 font-style: normal; font-variant-ligatures: normal; font-variant-caps: nor=
mal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left=
; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -we=
bkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(221=
, 221, 221); text-decoration-thickness: initial; text-decoration-style: ini=
tial; text-decoration-color: initial; display: inline !important; float: no=
ne;">Autodetect info: Fingerprint 55 (TTGO T-Beam (V0.7), 0.9" ***@***.***,22)<=
/span><br/><span style=3D"color: rgb(0, 0, 0); font-family: Arial, sans-ser=
if; font-size: medium; font-style: normal; font-variant-ligatures: normal; =
font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphan=
[... 77 lines cut ...]
isplay: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; h=
;dl9rdz/rdz_ttgo_sonde/issues/560/3181115535</span><span>@</span><span>gith=
ub</span><span>.</span><span>com&gt;</span></span></p>

<script type=3D"application/ld+json">[
{
"target": "https://github.com/dl9rdz/rdz_ttgo_sonde/issues/560#issuecomment=
-3181115535",
"url": "https://github.com/dl9rdz/rdz_ttgo_sonde/issues/560#issuecomment-31=
81115535",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
<br/>
<br/>
=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D Ende des Original Nachr=
ichtentextes =3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D=3D </i></span></body></ht=
ml>
------------12A0F41E7321ED663--

dl9rdz, 2025-08-12:
> ich dachte die PINS  21 22 sollte man wegen dem Powercontroller ab T-BEAM 1.0 nicht mehr nehmen.

Für das 0.96" OLED ist das ok, Powercontroller und OLED sind beides I2C, das geht parallel am gleichen Port.

Die größeren Displays sind SPI statt I2C, das geht dann nicht mit dem gleichen Port.

Wenn autodetect T-Beam v0.7 meint, dann steht evtl. Pin 13 bei Button2 drin, das muss geändert werden, falls an Pin 13 ein Display angeschlossen ist.

dl9rdz, 2025-08-12:
>   >  ich dachte die PINS 21 22 sollte man wegen dem Powercontroller ab T-BEAM 1.0 nicht mehr nehmen.
> Für das 0.96" OLED ist das ok, Powercontroller und OLED sind beides I2C, das geht parallel am gleichen Port.

Habe das nochmal überprüft... das ist nicht nur "ok", das OLED sollte beim T-Beam *immer* an 21, 22 angeschlossen werden. Intern wird der selbe Hardware-SPI-Controller für beides (Display, PMU) verwendet, daher ist das die einzige unterstützte Variante.